frawau/aioblescan #3: Eddystone URL encoding not being displayed correctly

ukBaz, 2017-10-30:
If the URL has a ".com/" in it then is not being decoded correctly from the packet.

For example the url of "https://makecode.com/#about" is coming out as follows:
```
Google Beacon {'tx_power': -10, 'rssi': -43, 'url': 'https://makecode\x00#about', 'mac address': 'f1:55:90:65:29:dc'}
```
This issue seems to be that the comparison is comparing a `int` with `bytes`

So changing from:
```python
 if x == b"\x00":
```

to:
```python
 if bytes([x]) == b"\x00":
```
Gives the following correct result:
```
Google Beacon {'rssi': -31, 'mac address': 'f1:55:90:65:29:dc', 'tx_power': -10, 'url': 'https://makecode.com/#about'}
```

I am sure there are other (and better?) ways of fixing this.

Do you have a preference for how this is fixed?

frawau, 2017-10-30:
Thanks for reporting that. I did not test that bit as you could see. I will publish a revised version soon. I do not know how your are using this library, but the new version will have a slight difference in the output of the RuuviWeather function: instead of  the keys "accel-x","accel-y" and "accel-z", it will have "accelerometer" and the value will be a 4-tuple (x,y,z,vector) with vector sqrt(x²+y²+z²). Hope that is ok.

frawau, 2017-10-30:
Pushed a new version. Let me know if it is OK so I can close this issue.

ukBaz, 2017-10-30:
Thanks for looking at this so quickly. It is working correctly now.

I do not have the RuuviTag. I am testing this library with the micro:bit . Programming of the micro:bit is done with the Bluetooth Eddystone capabilities at https://makecode.microbit.org/

closed 2017-10-30 by ukBaz