frawau/aioblescan #9: Scanner broken after connecting to a peripheral

Snevzor, 2018-01-30:
The scanner works fine. When connecting to a peripheral using [gatt-python](https://github.com/getsenic/gatt-python) for example the scanner stops, which is expected as the gatt-python discovery has messed up the HCI communication.

However after disconnecting and restarting the scanner it only shows advertisements from the last connected peripheral.

Restarting the HCI interface (hciconfig hci0 down/up) fixes this but I don't want to end up restarting the HCI device each time.
Also, this issue does not happen with hcitool lescan so I looked with strace at some differences between aioblescan and hcitool lescan:

**strace -o hciitool_strace hcitool lescan --passive --duplicates**

> socket(PF_BLUETOOTH, SOCK_RAW|SOCK_CLOEXEC, 1) = 3
> brk(NULL)                               = 0x555dc0bc5000
> brk(0x555dc0be6000)                     = 0x555dc0be6000
> ioctl(3, HCIGETDEVLIST, 0x555dc0bc5010) = 0
> ioctl(3, HCIGETDEVINFO, 0x7fff9fcc57e0) = 0
> close(3)                                = 0
> socket(PF_BLUETOOTH, SOCK_RAW|SOCK_CLOEXEC, 1) = 3
> bind(3, {sa_family=AF_BLUETOOTH, sa_data="\0\0\0\0\0\0\0\0\0\0\0\0\0\0"}, 6) = 0
> getsockopt(3, SOL_IP, IP_TTL, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", [16]) = 0
> setsockopt(3, SOL_IP, IP_TTL, "\20\0\0\0\1\300\0\0\0\0\0@\v \0\0", 16) = 0
> writev(3, [{"\1", 1}, {"\v \7", 3}, {"\0\20\0\20\0\0\0", 7}], 3) = 11
> poll([{fd=3, events=POLLIN}], 1, 10000) = 1 ([{fd=3, revents=POLLIN}])
> read(3, "\4\16\4\1\v \0", 260)          = 7
> setsockopt(3, SOL_IP, IP_TTL, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", 16) = 0
> getsockopt(3, SOL_IP, IP_TTL, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", [16]) = 0
> setsockopt(3, SOL_IP, IP_TTL, "\20\0\0\0\1\300\0\0\0\0\0@\f \0\0", 16) = 0
> writev(3, [{"\1", 1}, {"\f \2", 3}, {"\1\0", 2}], 3) = 6
> poll([{fd=3, events=POLLIN}], 1, 10000) = 1 ([{fd=3, revents=POLLIN}])
> read(3, "\4\16\4\1\f \0", 260)          = 7
> setsockopt(3, SOL_IP, IP_TTL, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", 16) = 0
> fstat(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 10), ...}) = 0
> write(1, "LE Scan ...\n", 12)           = 12
> getsockopt(3, SOL_IP, IP_TTL, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", [16]) = 0
> setsockopt(3, SOL_IP, IP_TTL, "\20\0\0\0\0\0\0\0\0\0\0@\0\0\0\0", 16) = 0
> rt_sigaction(SIGINT, {0x555dc0073d40, [], SA_RESTORER|SA_NOCLDSTOP, 0x7fa6ce377390}, NULL, 8) = 0
> read(3, "\4>\35\2\1\0\0\1\21\316\370\346\240\21\2\1\6\5\tlynx\7\377\377\377\1\1\1\24\264", 260) = 32
> write(1, "A0:E6:F8:CE:11:01 lynx\n", 23) = 23

**sudo strace -o scanner_strace python3 ./LAH-547/scanner.py**

> socket(PF_BLUETOOTH, SOCK_RAW|SOCK_CLOEXEC, 1) = 6
> ioctl(6, FIONBIO, [1])                  = 0
> setsockopt(6, SOL_IP, IP_TTL, "\377\377\377\377\377\377\377\377\377\377\377\377\0\0\0\0", 16) = 0
> bind(6, {sa_family=AF_BLUETOOTH, sa_data="\0\0\0\0\0\0\0\0\0\0\0\0\0\0"}, 6) = 0
> epoll_wait(3, [], 1, 0)                 = 0
> ioctl(6, FIONBIO, [1])                  = 0
> getsockname(6, {sa_family=AF_BLUETOOTH, sa_data="\0\0\0\0\0\0\0\0\0\0\0\0\0\0"}, [6]) = 0
> getpeername(6, 0x7ffc28e85b60, 0x7ffc28e85b5c) = ­1 EOPNOTSUPP (Operation not supported)
> epoll_wait(3, [], 1, 0)                 = 0
> write(1, "CONNECTION MADE\n", 16)       = 16
> epoll_ctl(3, EPOLL_CTL_ADD, 6, {EPOLLIN, {u32=6, u64=18115095873162575878}}) = 0
> epoll_wait(3, [], 2, 0)                 = 0
> epoll_wait(3, [], 2, 0)                 = 0
> sendto(6, "\1\f \2\1\0", 6, 0, NULL, 0) = 6
> stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=2970, ...}) = 0
> write(1, "29­01­2018 14:42:11.692 DEBUG [s"..., 63) = 63
> epoll_wait(3, [{EPOLLIN, {u32=6, u64=18115095873162575878}}], 2, ­1000) = 1
> mmap(NULL, 266240, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, ­1, 0) = 0x7f65f84b3000
> recvfrom(6, "\4\16\4\1\f \0", 262144, 0, NULL, NULL) = 7
> mremap(0x7f65f84b3000, 266240, 4096, MREMAP_MAYMOVE) = 0x7f65f84b3000
> munmap(0x7f65f84b3000, 4096)            = 0
> epoll_wait(3, [{EPOLLIN, {u32=6, u64=18115095873162575878}}], 2, ­1000) = 1
> mmap(NULL, 266240, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, ­1, 0) = 0x7f65f84b3000
> recvfrom(6, "\4>\32\2\1\0\0\4\22\257\370\346\240\16\2\1\6\3\2\200\252\6\377\r\0\3\0\0\264", 262144, 0, 
> NULL, NULL) = 29
> mremap(0x7f65f84b3000, 266240, 4096, MREMAP_MAYMOVE) = 0x7f65f84b3000
> munmap(0x7f65f84b3000, 4096)            = 0
> epoll_wait(3, [{EPOLLIN, {u32=6, u64=18115095873162575878}}], 2, ­1000) = 1
> mmap(NULL, 266240, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, ­1, 0) = 0x7f65f84b3000
> recvfrom(6, "\4>!\2\1\0\0\5\v\243\16lT\25\2\1\6\5\tLynx\v\377\377\377\2\1\0\0\0"..., 262144, 0, NULL, NULL) 
> = 36
> 

So it appears that hcitool is setting more commands after opening the socket? aioblescan should also take into account that other software can mess up HCI settings.

I hope someone can help me with this.

Best regards,
Sven

Snevzor, 2018-01-31:
Also the issue is fixed when scanning briefly with hcitool. So hcitool appears to be applying some settings by default that aioblescan is not.

frawau, 2018-01-31:
Hi,

Thanks for reporting this. I am a bit busy right now, but I'll see what I can do.

Regards,
	François

On Tuesday, 30 January 2018 23:06:17 +07 Snevzor wrote:

The scanner works fine. When connecting to a peripheral using gatt-python[1] for 
example the scanner stops, which is expected as the gatt-python discovery has messed up 
the HCI communication.
However after disconnecting and restarting the scanner it only shows advertisements 
from the last connected peripheral.
Restarting the HCI interface (hciconfig hci0 down/up) fixes this but I don't want to end up 
restarting the HCI device each time.Also, this issue does not happen with hcitool lescan so 
I looked with strace at some differences between aioblescan and hcitool lescan:
*strace -o hciitool_strace hcitool lescan --passive --duplicates*
socket(PF_BLUETOOTH, SOCK_RAW|
SOCK_CLOEXEC, 1) = 3brk(NULL)                               = 0x555dc0bc5000brk(0x555dc0be6000)  
                   = 0x555dc0be6000ioctl(3, HCIGETDEVLIST, 0x555dc0bc5010) = 0ioctl(3, HCIGETD
EVINFO, 0x7fff9fcc57e0) = 0close(3)                                = 0socket(PF_BLUETOOTH, SOCK_RA
W|
SOCK_CLOEXEC, 1) = 3bind(3, {sa_family=AF_BLUETOOTH, sa_data="\0\0\0\0\0\0\0\0\0\0\0\
0\0\0"}, 6) = 0getsockopt(3, SOL_IP, IP_TTL, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", [16]) = 0sets
ockopt(3, SOL_IP, IP_TTL, "\20\0\0\0\1\300\0\0\0\0\0@
\v \0\0", 16) = 0writev(3, [{"\1", 1}, {"\v \7", 3}, {"\0\20\0\20\0\0\0", 7}], 3) = 11poll([{fd=3, eve
nts=POLLIN}], 1, 10000) = 1 ([{fd=3, revents=POLLIN}])read(3, "\4\16\4\1\v \0", 260)          = 
7setsockopt(3, SOL_IP, IP_TTL, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", 16) = 0getsockopt(3, SOL
_IP, IP_TTL, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", [16]) = 0setsockopt(3, SOL_IP, IP_TTL, "\20\0
\0\0\1\300\0\0\0\0\0@
\f \0\0", 16) = 0writev(3, [{"\1", 1}, {"\f \2", 3}, {"\1\0", 2}], 3) = 6poll([{fd=3, events=POLLIN}],
 1, 10000) = 1 ([{fd=3, revents=POLLIN}])read(3, "\4\16\4\1\f \0", 260)          = 7setsockopt(3, 
SOL_IP, IP_TTL, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", 16) = 0fstat(1, {st_mode=S_IFCHR|
0620, st_rdev=makedev(136, 10), ...}) = 0write(1, "LE Scan ...
\n", 12)           = 12getsockopt(3, SOL_IP, IP_TTL, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", [16]) = 0
setsockopt(3, SOL_IP, IP_TTL, "\20\0\0\0\0\0\0\0\0\0\0@
\0\0\0\0", 16) = 0rt_sigaction(SIGINT, {0x555dc0073d40, [], SA_RESTORER|
SA_NOCLDSTOP, 0x7fa6ce377390}, NULL, 8) = 0read(3, "\4>
\35\2\1\0\0\1\21\316\370\346\240\21\2\1\6\5\tlynx
\7\377\377\377\1\1\1\24\264", 260) = 32write(1, "A0:E6:F8:CE:11:01 lynx\n", 23) = 23
*sudo strace -o scanner_strace python3 ./LAH-547/scanner.py*
socket(PF_BLUETOOTH, SOCK_RAW|
SOCK_CLOEXEC, 1) = 6ioctl(6, FIONBIO, [1])                  = 0setsockopt(6, SOL_IP, IP_TTL, "\377
\377\377\377\377\377\377\377\377\377\377\377\0\0\0\0", 16) = 0bind(6, {sa_family=AF_BL
UETOOTH, sa_data="\0\0\0\0\0\0\0\0\0\0\0\0\0\0"}, 6) = 0epoll_wait(3, [], 1, 0)                 = 0i
octl(6, FIONBIO, [1])                  = 0getsockname(6, {sa_family=AF_BLUETOOTH, sa_data="\0\
0\0\0\0\0\0\0\0\0\0\0\0\0"}, [6]) = 0getpeername(6, 0x7ffc28e85b60, 0x7ffc28e85b5c) = ­
1 EOPNOTSUPP (Operation not supported)epoll_wait(3, [], 1, 0)                 = 0write(1, "CONN
ECTION MADE
\n", 16)       = 16epoll_ctl(3, EPOLL_CTL_ADD, 6, {EPOLLIN, {u32=6, u64=1811509587316257
5878}}) = 0epoll_wait(3, [], 2, 0)                 = 0epoll_wait(3, [], 2, 0)                 = 0sendto(6, "\1\f 
\2\1\0", 6, 0, NULL, 0) = 6stat("/etc/localtime", {st_mode=S_IFREG|
0644, st_size=2970, ...}) = 0write(1, "29­01­
2018 14:42:11.692 DEBUG [s"..., 63) = 63epoll_wait(3, [{EPOLLIN, {u32=6, u64=18115095873
162575878}}], 2, ­1000) = 1mmap(NULL, 266240, PROT_READ|PROT_WRITE, MAP_PRIVATE|
MAP_ANONYMOUS, ­
1, 0) = 0x7f65f84b3000recvfrom(6, "\4\16\4\1\f \0", 262144, 0, NULL, NULL) = 7mremap(0x7
f65f84b3000, 266240, 4096, MREMAP_MAYMOVE) = 0x7f65f84b3000munmap(0x7f65f84b3
000, 4096)            = 0epoll_wait(3, [{EPOLLIN, {u32=6, u64=18115095873162575878}}], 2, ­
1000) = 1mmap(NULL, 266240, PROT_READ|PROT_WRITE, MAP_PRIVATE|
MAP_ANONYMOUS, ­1, 0) = 0x7f65f84b3000recvfrom(6, "\4>
\32\2\1\0\0\4\22\257\370\346\240\16\2\1\6\3\2\200\252\6\377\r
\0\3\0\0\264", 262144, 0, NULL, NULL) = 29mremap(0x7f65f84b3000, 266240, 4096, MREM
AP_MAYMOVE) = 0x7f65f84b3000munmap(0x7f65f84b3000, 4096)            = 0epoll_wait(3, [{
EPOLLIN, {u32=6, u64=18115095873162575878}}], 2, ­
1000) = 1mmap(NULL, 266240, PROT_READ|PROT_WRITE, MAP_PRIVATE|
MAP_ANONYMOUS, ­1, 0) = 0x7f65f84b3000recvfrom(6, "\4>!\2\1\0\0\5\v\243\16lT
\25\2\1\6\5\tLynx\v\377\377\377\2\1\0\0\0"..., 262144, 0, NULL, NULL) = 36
So it appears that hcitool is setting more commands after opening the socket? aioblescan 
should also take into account that other software can mess up HCI settings.
I hope someone can help me with this.
Best regards,Sven 

--------
[1] https://github.com/getsenic/gatt-python
[2] https://github.com/frawau/aioblescan/issues/9
AORvqcBWVIj7pDnWDS8dB_jAdBB2C1CRks5tPz35gaJpZM4RylGh
[4] https://github.com/notifications/beacon/
AORvqcS4eIAicFxvAloRBHUppQLqNbp0ks5tPz35gaJpZM4RylGh.gif

frawau, 2018-01-31:
Do you mind describing your environment. Especially the hardware you are using.

Snevzor, 2018-01-31:
Hello @frawau, thanks for responding!

I've just found what was missing. There were no 'set scan params' but luckily you've added this in commit 58ee94cf4cc4ddc2ecf4b6ef2136edc61c0b0a9c (Also added support for HCI_Cmd_LE_Set_Scan_Params. Not sure how useful that is..)!

So in BLEScanRequester I've updated:
```
def connection_made(self, transport):
        self.transport = transport
        self.transport.write(command.encode()
```
To:
```
def connection_made(self, transport):
        self.transport = transport
        command=HCI_Cmd_LE_Set_Scan_Params()
        self.transport.write(command.encode()
```

And now it doesn't matter anymore if another piece of software has messed up the parameters.
Your default values seem to match those of hcitool lescan --duplicates :+1: 

(In case of gatt-python there is a discovery just before connection with a filter on MAC address I presume.)

frawau, 2018-02-01:
Thanks @Snevzor 

Will you be submitting a PR or should I go ahead and commit your change?

frawau, 2018-02-01:
Thanks for your help.

closed 2018-02-01 by frawau