frawau/aioblescan #12: Decoding issue for temperaturs below 0C?

Anppa, 2018-03-21:
python 3.6.4, aioblescan 0.2.0 from pypi, running 'python -m aioblescan -r' for a ruuvi tag cooling in the freezer:

```
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': 0.3, 'pressure': 100463, 'accelerometer': (-8, 8, 1044, 1044.0613008822806), 'voltage': 3097}
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': 0.25, 'pressure': 100464, 'accelerometer': (-8, 16, 1036, 1036.154428644688), 'voltage': 3091}
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': 0.22, 'pressure': 100469, 'accelerometer': (-4, 8, 1032, 1032.0387589620848), 'voltage': 3085}
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': 0.14, 'pressure': 100458, 'accelerometer': (0, 12, 1036, 1036.0694957385822), 'voltage': 3079}
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': 0.12, 'pressure': 100464, 'accelerometer': (-4, 8, 1032, 1032.0387589620848), 'voltage': 3085}
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': 0.05, 'pressure': 100462, 'accelerometer': (-4, 8, 1036, 1036.03860931917), 'voltage': 3091}
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': 0.0, 'pressure': 100460, 'accelerometer': (-4, 8, 1032, 1032.0387589620848), 'voltage': 3085}
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': -127.96, 'pressure': 100464, 'accelerometer': (-8, 12, 1036, 1036.1003812372621), 'voltage': 3097}
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': -127.89, 'pressure': 100465, 'accelerometer': (-4, 12, 1036, 1036.0772171995677), 'voltage': 3079}
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': -127.68, 'pressure': 100463, 'accelerometer': (-8, 12, 1040, 1040.09999519277), 'voltage': 3091}
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': -127.64, 'pressure': 100462, 'accelerometer': (-8, 8, 1036, 1036.0617742200511), 'voltage': 3085}
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': -127.56, 'pressure': 100464, 'accelerometer': (-8, 8, 1036, 1036.0617742200511), 'voltage': 3091}
Weather info {'mac address': 'f0:a2:ce:24:60:45', 'humidity': 21.5, 'temperature': -127.54, 'pressure': 100460, 'accelerometer': (-8, 8, 1032, 1032.0620136406533), 'voltage': 3091}
```

An unsigned/signed issue? Android app reports the temperature correctly. I can dig more into this, I just thought I'll report here first.

frawau, 2018-03-22:
Ooops looks like that... I live in Bangkok, I don't get below zero temperature, unless it's in a freezer ';)

samuelphy, 2018-03-22:
I have a fix for this, it's not two complement, it just has the first bit
as sign. I am away from the computer now but can make a pull request in a
couple of days.
tors 22 mars 2018 kl. 04:57 skrev François Wautier <notifications@github.com

>:

> Ooops looks like that... I live in Bangkok, I don't get below zero
> temperature, unless it's in a freezer ';)
-- 

*Elphy AB*

Hålekärrsg. 8B, 41466 Göteborg
Phone: +46 (0)72-017 49 71
www.elphy.se | info@elphy.se

closed 2018-03-29 by frawau

Anppa, 2018-03-30:
\o/ thanks!

Watchfox, 2018-11-29:
> 
> I have a fix for this, it's not two complement, it just has the first bit as sign. I am away from the computer now but can make a pull request in a couple of days.

It's actually not quite fixed.

```
result["temperature"]=s8(val[2])
result["temperature"]+=val[3]/100.0
```

When it comes to the decimal part, it is '+' if the temperature is positive but should be '-' when negative. See the example here: https://github.com/ruuvi/ruuvi-sensor-protocols (under Data Format 3 Protocol Specification)

Anppa, 2018-11-30:
That would explain

[image]

samuelphy, 2018-11-30:
Yes, I see the mistake.
I can fix it this weekend, that is if nobody else wants to.

frawau, 2018-11-30:
Please do.

I use my 3 Ruuvi, but I won't get negative temperatures in Bangkok.

Putting one in the freezer would mess up my data.

So I cannot test.

Cheers,
    François

On Friday, 30 November 2018 18:57:36 +07 Samuel Vestlin wrote:

Yes, I see the mistake.I can fix it this weekend, that is if nobody else wants to. 

--------
[1] https://github.com/frawau/aioblescan/issues/12#issuecomment-443182507
AORvqUG3GdkwgbKz_SEEb2FKKIe5ioIJks5u0R0wgaJpZM4S1d4Y
[3] https://github.com/notifications/beacon/
AORvqZJ7eQdRCEN5TUsl_S1HTbnKHSFJks5u0R0wgaJpZM4S1d4Y.gif

ukBaz, 2018-11-30:
Is it possible to write a unit test for RuuviWeather().decode(ev)?

commit 20a99645 (2018-03-29): Fixed temperature sign conversion for Ruuvitags which closes #12. Also move RSSI measurement to make it accessible for raw data