frawau/aioblescan #18: How to run without install

UweBonnes, 2018-09-27:
Hello,
after some fiddling around with opensuse 42.3 python3 not having AF_BLUETOOTH support, I can now run " python3 -m aioblescan -r"  and see the Ruuvi devices. However they are intersparsed with 
url oops....
HCI Event:
    code:
        3e
         ...
output.
So I clone the repo to play with the code.  However running
aioblescan/aioblescan # python3 aioblescan.py  -r
does not print out anything. what do I miss?

Thanks

frawau, 2018-09-28:
Hi,

Not sure what is happening, I would need the actual  output to know,. but I suspect that you are not using the right mode on the Ruuvi

By default, the Ruuvi broadcast EddyStone messages. 
 You should use the "-e" option to decode those.

By pushing the button on the Ruuvi tag, you can change the firmware mode and get it to send the info decoded by the "-r" option.

Now... it is also possible that the format changed and that I am not aware of it.

You can still run the module with no option and see  if there is any error coming out. This way we can figure out if it is a scanning issue or a decoding issue.

Regards,
    François

UweBonnes, 2018-10-04:
Thanks for your answer. Recent Ruuvis run the weatherstation firmware as default.  I now use https://github.com/ttu/ruuvitag-sensor. The latter does not need the AF_BLUETOOTH socket option. The AF_BLUETOOTH option s not yet available in Opensuse Python and providing it is a problem for the package mantainer due to circular dependancies.

closed 2018-10-04 by UweBonnes