frawau/aioblescan #22: Ruuvi data not identified to ruuvi data

fuggah, 2019-03-30:
I have a recently purchased Ruuvi device. Get only raw data with that although using -r switch.

length = 1
data = (31) b'\x03\x01\x1a\x8b\xdd\xb2\xe7\xc0\x15\x02\x01\x06\x11\xff\x99\x04\x03\x98\x02\x11\xc3\xf7\xff\x14\x03\xf1\x00I\x0b\xdd\xa3'
datalength = 21
datalength = 18
Adv Report:
    ev type:
        no connection adv
    addr type:
        random
    peer:
        c0:e7:b2:dd:8b:1a
    length:
        21
    flags:
        Simul LE - BR/EDR (Host): False
        Simul LE - BR/EDR (Control.): False
        BR/EDR Not Supported: False
        LE General Disc.: True
        LE Limited Disc.: True
    Manufacturer Specific Data:
        Manufacturer ID:
            1177
        Payload:
            03:98:02:11:c3:f7:ff:14:03:f1:00:49:0b:dd
    rssi:
        -93

frawau, 2019-03-31:
Hi,

Are you sure that is a Ruuvi tag? 
What mode are you using? What firmware version?What data are you expecting?  The data seems to short for either URL or manufacturer specific data.

Can you run  
      python3 -m aioblescan -R

So I can see what aioblescan is trying to decode

fuggah, 2019-03-31:
Hi @frawau 

Yes, it is absolutely a Ruuvi tag.
I added two files; one from aioblescan like you suggested and an other one from ttu/ruuvitag_sensor. Hope that helps.

[aioblescan.log](https://github.com/frawau/aioblescan/files/3026304/aioblescan.log)
[ruuvitag.log](https://github.com/frawau/aioblescan/files/3026305/ruuvitag.log)

frawau, 2019-04-01:
Hi,

I am a bit puzzled. I ran your raw data through the decoder and it worked...

How did you install aioblescan? Could you try cloning the github repo and install it from there?
Did you try running aioblescan with the Ruuvi tag in "URL" mode (Usually the default mode after power cycling)

The only difference I see between your data and mine is some extra null bytes in my packets

petri, 2019-11-24:
@fuggah what is your __exact__ command line? If you don't limit the MAC addresses using the -m switch, you get all the advertisments from all the devices around you. There may be many.

This tricked me at first - there was so much information running thru my terminal that I did not even notice the ruuvi entries generated by -r switch.

@frawau is this intentional behavior? I thought that by giving a -r switch I would _only_ get the ruuvi tag entries.

fuggah, 2019-11-26:
@petri , well it is so long ago this happened that I needed to spend some time recalling. 
The command line was 'python3 -m aioblescan -m C0:E7:B2:DD:8B:1A -r 2>&1 >withMAC.log'

I also tried with 'python3 -m aioblescan -r 2>&1 >withoutMAC.log'

No matchstick.
The MAC address is copied from the working framework 'ruuvitag-logger' which is now running my cron.hourly and saving the data into a sqlite database.

frawau, 2019-11-27:
@petri Yes it is how it is intended to work. The '-r' flag requests to try to decode the data as a Ruuvi tag data. To limit the packet displayed, use the mac filter.

fuggah, 2019-11-27:
Ok, so the result is that for some reason it cannot decode it as a Ruuvi tag. I need to make some comparison with ruuvitag-logger and aioblescan data. I bet there's a small difference how the data is decoded

frawau, 2020-02-02:
Hi 

You could have been using RAWv2. Check with the version I released yesterday. 0.2.6

fuggah, 2020-02-13:
Yo, sorry about the delay. Installed 0.2.6 and it seems to work now. Data is identified to Ruuvi data.

closed 2020-02-13 by fuggah