frawau/aioblescan #23: Suggestion: run without root access

cHemingway, 2019-06-21:
I have been looking into how to run this module without using sudo, and I think I have an option, but have not tried it myself, so apologies if this is a wild goose chase:

To access the HCI socket, you apparently need [cap_net_raw and cap_net_admin](https://unix.stackexchange.com/a/182559) capabilities to not run as root. Unfortunately, these capabilities would need to be set per binary (e.g. python interpreter) which causes issues with interpreted languages as every file executed by the interpreter would have these privileges.

The library [rawsocket](https://github.com/mwalle/rawsocket) improves the security of this slightly by using a helper binary with these capabilities to create a socket with cap_net_raw and then passes the socket back. Unfortunately, this library is Python2 only and does not add the CAP_NET_ADMIN capability, but maybe it could be forked and modified to?

frawau, 2019-06-24:
Hi,

Thanks for the suggestion. I am not sure what your use case is.

As I see it, my current use case are mostly home automation  on Raspberry Pi or similar platform. Not sure if rawsocket is worth the effort. It is only Linux (fine I don't care about Windows)  and sudo has all the flexibility needed AFAIC.

Cheers,
    François.

cHemingway, 2019-06-24:
Hi,

My use case would be home automation as well, not needing the service to run as root adds security.

Other bluetooth components (e.g. gatt-python) use the BlueZ APIs over dbus and so can avoid needing root that way, but as your library needs HCI access, this sockets trick is needed.

Of course, the chance of my raspberry pi being attacked is quite low, and I appreciate this is quite a complex solution to mitigate that small risk.

No worries if you aren't interested, this is a great library already, thanks for your work on it.

Cheers,
Chris

closed 2019-06-24 by cHemingway