frawau/aioblescan #24: Can't detect EddyStone URL advertisement with 5.0 kernel

yphus, 2019-07-22:
aioblescan can't detect EddyStone URL advertisement with 5.0 kernel.

We could see "Bluetooth: hci0: advertising data len corrected" is received in dmesg, however, aioblescan can't detect the advertisement.

See the code below:

`  class HCI_LE_Meta_Event(Packet):`
`    def decode(self,data):`
`      ...`
`      if code.val==b"\x02":`
`          ev=RepeatedField("Adv Report", HCI_LEM_Adv_Report)`
`         data=ev.decode(data)`
`          self.payload.append(ev)`
`      else:`
`          ...`
`

aioblescan supports only the "legacy" LE Advertising report(0x2).

Need to implement corresponding handling code based on BLUETOOTH CORE SPECIFICATION Version 5.1, section 7.7.65.13 LE Extended Advertising Report event (code 0xd).

frawau, 2019-07-25:
Sorry, I am currently vacationing in France. Will have a look at it when I am back home, mid Aug.

yphus, 2019-07-25:
Thanks for the reply (I'll do the same next week)
Enjoy your holidays.

Sylvain PINEAU

yphus, 2019-10-22:
Hi, have you been able to reproduce the issue? I can help providing more data from the system where it happens.

frawau, 2019-10-25:
Sorry, I have not had time yet. I am about to move to a new house and it is taking more of my time than I expected. But please send me the setup you use to reproduce the problem.

Cheers
    François

vicamo, 2019-11-25:
Proposed #27 to fix this issue.

closed 2019-11-27 by frawau

vicamo, 2019-11-27:
@frawau will we have a new release for this? I've tried this against EddyStone beacon I got, but maybe you have also other hw to test against as well.

frawau, 2019-11-27:
I will check with my Ruuvi devices first. Not sure when but soon I hope

commit 25950919 (2019-11-27): Support extended adv report

See BLUETOOTH CORE SPECIFICATION Version 5.1, Vol 2: Core System
Package, Part E Host Controller Interface Functional Specification,
section 7.7.65.13 LE Extended Advertising Report event.

Closes: #24