frawau/aioblescan #25: Scan parameters have a huge impact on Wi-Fi performance on single chip Wi-Fi/Bluetooth radio's

pruckebusch, 2019-10-31:
We noticed that the current settings have a huge impact on the Wi-Fi performance on single chip Wi-Fi Bluetooth radio's. Using these settings, Wi-Fi throughput drops significantly and stations have problems connecting with their AP.

As we need both Wi-Fi and Bluetooth (we have a healthcare solution with bluetooth peripherals and fixed nodes, our nodes use Wi-Fi as backbone network) any recommendation on an alternative setting would be appreciated.

We also noticed that the default value for window used by the HCI_Cmd_LE_Set_Scan_Params function is not correct (i.e. should be 10).

frawau, 2019-11-02:
Hi,  

Sorry, I am currently using this on a RPi2 with an added bluetooth dongle. So I am not having your issue.  What chipset are you using?

pruckebusch, 2019-11-04:
Hi,

I am using the cypress (formerly broadcom) cyw4343w chipset.
I think this chip also used on RPi3.

frawau, 2019-11-04:
I think I have a RPi Zero W lying around,  I'll see if I can test things out, it has a cyw43438 chipset, that should not be too different. End of the week probably...

I would advise you to contact Cypress to see if they have any info on your issue.

pruckebusch, 2019-11-05:
Ok, thanks.
We are already in contact with Cypress to see if we can fine tune Wi-Fi/Bluetooth coexistence.
I'll post a comment here if we come up with something.

I also wondered why the default value for window is 750.

frawau, 2019-11-06:
Yes please, let me know if Cypress get back to you with something

As for the window value, it looks like an error. In any case the value used is min(interval,window) so that high value is unlikely to be used

DASOSiot, 2020-06-03:
I went deep into this issue with heavy scanning of BLE packages using a RPi 3 (power micro board). So this issue goes from basic boards to power boards as RPi 3.

SOLUTION: after countless attempts, we tested several 5 USD BLUETOOTH USB DONGLE (here it goes deep, because some times you have to compile & update packages related to bluetooth https://github.com/DASOSiot/desk/blob/master/scripts/steps/09.sh ). 

Use the board only as WiFi. Build your Bluetooth apps using the dongle. We gain perfect connectivity and very granular non intrusive BLE scanning.

closed 2023-01-08 by frawau