frawau/aioblescan #28: Enable active scan and receive type 4 scan response packets

deepcoder, 2020-04-30:
Hello,
It appears that you have the ability to enable active scanning in HCI_Cmd_LE_Set_Scan_Params .
Have you tried this with your code and decoded these scan response packets?

Thanks in advance,
Dave

REF:
Scan Response

In addition to the 31 bytes of advertising data, the device doing the scanning can request the scan response data. This does not require authentication, and does not require making a connection. The scan response data is an additional 31 bytes of data the peripheral can return to the scanning device, though it takes an extra set of packets to and from the peripheral to request and receive it.

frawau, 2020-05-01:
Hi,

No I have not tried that. That bit of code (HCI_Cmd_LE_Set_Scan_Params) was lifted from scapy. 

Not sure I have a device that would reply to that type of request.... Also not sure if the extra packet has to be actively sent to the peripheral or if the lower level would take care of that...

Cheers,
    François

ukBaz, 2020-05-23:
Hi,

I've already added the Blue Maestro device format to this library which also has extra information in the scan response so I would be interested in this also.

I've done a HCI trace using the btmon tool. I started the scan with bluetoothctl.
This report shows that the event types are ADV_IND (0x00) and SCAN_RSP (0x04)

```
> HCI Event: LE Meta Event (0x3e) plen 43                  #63 [hci0] 35.867823
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Random (0x01)
        Address: DC:76:F7:E1:62:E0 (Static)
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Company: Blue Maestro Limited (307)
          Data: 1b640e100006011601be27ba0100
        Name (complete): DC76F7E1
        RSSI: -69 dBm (0xbb)

> HCI Event: LE Meta Event (0x3e) plen 41                  #64 [hci0] 35.870235
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scan response - SCAN_RSP (0x04)
        Address type: Random (0x01)
        Address: DC:76:F7:E1:62:E0 (Static)
        Data length: 29
        Company: Blue Maestro Limited (307)
          Data: 27ca27c227b900000124020d010401be011001e70000000000
        RSSI: -70 dBm (0xba)
```

@deepcoder have you confirmed if both of these reports appear in the HCI trace when the scan is launched with aioblescan?
I am trying to workout if aioblescan filters them out somewhere along the way...

ukBaz, 2020-05-24:
I have done a little bit more research on this. Running a btmon HCI trace I have confirmed that aioblescan does not cause the SCAN_RSP (0x04) event types to happen by default

Changing the `scan_type` default from `0x0` to `0x1` in the below line does cause the SCAN_RSP to be included in the HCI trace:
https://github.com/frawau/aioblescan/blob/ac3fbcb414573a47786903f9e17b169107c003c0/aioblescan/aioblescan.py#L799-L828

The only place this gets called from is here where there is no option to change the parameters:
https://github.com/frawau/aioblescan/blob/ac3fbcb414573a47786903f9e17b169107c003c0/aioblescan/aioblescan.py#L1321-L1324

In hindsight, that looks like it was @deepcoder initial point, just took me a while to get up to speed.
I think it would be useful to have this as a choice in the library. Not sure the best way of doing that so asking the question here.

@frawau: If you are interested, I could look at getting a [Blue Maestro Pebble](https://bluemaestro.com/products/product-details/pebble-bluetooth-environment-monitor) shipped to you