frawau/aioblescan #32: Use double quotes for JSON output

faceless2, 2021-01-06:
Minor suggestion - JSON requires double-quotes to be used for strings, so it would make sense to change the output of the various plugins to use these rather than single quotes - eg `{"rssi": -64, "pressure": 100300, ...}`. It's a little thing but makes it easier to parse the output of aioblescan with other tools

dankozlowski, 2023-01-13:
It may be worth opening your own PR for this one.  Maybe use a command param or something to make this output change?

frawau, 2023-01-14:
The thing is, this is mostly a library with a little utility thrown in so people can play with it.

Getting that json output is quite trivial. It is now done, but I don;t feel it justifies publishing a new release.

Will be included in the next release.  If you need it now, install from sources.

closed 2023-01-14 by frawau