frawau/aioblescan #33: aioblescan-0.2.7.tar.gz hosted on pypi.org doesn't contain README.md file.

gblach, 2021-02-25:
When I tried to build aioblescan from source downloaded from https://files.pythonhosted.org/packages/78/2c/c09d8f83ed6c2ca6c4f803f2e5c349698eb78b82c8d5e9302a8eed1a7480/aioblescan-0.2.7.tar.gz

it fails because file README.md is missing.

    $ python3 setup.py
    Traceback (most recent call last):
    File "/Users/magik/Downloads/aioblescan-0.2.7/setup.py", line 7, in <module> with open("README.md", "r") as fh:
    FileNotFoundError: [Errno 2] No such file or directory: 'README.md'

frawau, 2021-02-25:
Hi, 

I see ...  Will be corrected once I up the version. In the meantime , use
      pip install aioblescan

If you really must use setup.py, you could edit it and replace README.md by README.rst, or get the sources from github

Also, on Linux, 
       touch README.md
Before you run setup.py

frawau, 2021-11-03:
No longer an issue AFAICT

closed 2021-11-03 by frawau