frawau/aioblescan #36: Error on wrong data

Ernst79, 2021-07-25:
Hi @frawau 

We have received an issue from one of the users of BLE monitor (@LEJOUI) , in which we use your aioblescan, as you probably know. 

https://github.com/custom-components/ble_monitor/issues/433

We did some debugging and found out that this user (sometimes) receives strange data, which causes an error in aioblescan. 

```
Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:87 
First occurred: 18:34:47 (2 occurrences) 
Last logged: 18:42:18

Error doing job: Fatal error: protocol.data_received() call failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/selector_events.py", line 870, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 1840, in data_received
    extra_data = ev.decode(packet)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 1303, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 1350, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 833, in decode
    data = field.decode(data)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 1379, in decode
    data = ad.decode(data)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 1646, in decode
    val.decode(data[: length.val - len(type)])
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 664, in decode
    data = mynbyte.decode(data)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 623, in decode
    self.val = unpack(">%ds" % self.length, data[: self.length])[0][::-1]
struct.error: unpack requires a buffer of 4 bytes
```

We tried to cache some of the ble advertisements that caused the errors. These are some of the errors that are causing similar errors like above. 

```
failing data: 043e4e02020300ae87bb59af011918ff4c000713010e2071aa37320010006164ff000000000000ae0001efa3f561b3761f02011a1bff4c000c0e00afe8e025efb044dbb3995e35da10065b1e5c501877bc
failing data: 043e4602020000eef0875da17c1109ff89004591624583320302fefe020104a80400eef0875da17c1f0309474e1aff89005504001f0111001a0d11064ad365128aac94fc68448160a8
failing data: 043e3c020202015b698bbad1471c03039ffe17169ffe0000000000000000000000000000000000000000bc04015b698bbad1470a09ffe000b213ca8d9b0fba
failing data: 043e2702020001962fda3146681102011a020a0c0aff4c0010050318f88918ba0401962fda31466800ba
```

I tried to analyze the last advertisement. 

```
--------------------------------------------------------------------------------------------------------------------------------------------------------------
HCI  Evt Len Sub Num Evt  Peer -------MAC-------   Len  Len Type Val    Len Type Val   Len Type -------Val---------------- RSSI--?????????????????????????????  
type code    evt rep type addr                              flag            flag           flag
--------------------------------------------------------------------------------------------------------------------------------------------------------------
 04   3e 27  02  02   00   01  96 2f da 31 46 68   11   02   01  1a     02   0a  0c     0a  ff  4c 00 10 05 03 18 f8 89 18   ba   04 01 96 2f da 31 46 68 00 ba
```

There is something strange at the end. Normally, after the mac, you would have something like the total remaining length excl RSSI (in this case 11 in hex = 17 bytes). In the payload, this is split up in subparts with length, type and value:

`02 01 1a` (length 2, type 01, value 0a)
`02 0a 0c` (length 2, type 0a, value 0c)
`0a ff 4c 00 10 05 03 18 f8 89 18`, length 10, type ff, value the remaining 9 bytes
Next, you would get the RSSI value (`ba`). So far, so good. 

But after that, there are additional bytes
`04 01 96 2f da 31 46 68 00 ba`

But I can't make sense of it, length isn't 4, I do see the same RSSI at the end (`ba`). But this does not make sense to me. I expect that aioblescan is failing on this part as is continues to decode data. 

Error

```
  File "C:\Users\903494\Documents\decryptor\aioblescan_ext.py", line 627, in decode
    self.val = unpack(">%ds" % self.length, data[: self.length])[0][::-1]
struct.error: unpack requires a buffer of 4 bytes
```

It seems to fail, as in the AD_Structure class, it continues to decode this last part and than runs into trouble, as it tries do decode things with the wrong number of bytes or even no bytes.

Do you have any idea what could be wrong and how this can be fixed (or skip these advertisements?)

frawau, 2021-07-26:
Hi,
Well, those extra octets do contain the MAC address again.

Beside that.. it does seem like garbage even though they seem to be included in the full length of the packet.

I noticed that user of yours  is using  a custom ATC1441 firmware. I would suspect something wrong with that. He does mention that it was working "in version 3.5 but not in version 4.0"  Not sure what version he is talking about... 

Are you using the atcmithermometer plugin?  That one support the custom mode data. Maybe that could be used..

My solution would be for the app to catch the exception and throw the data. As you have noted the end of the message looks like garbage, so solving that in the protocol, would require some ugly hackery... probably not a good idea. 

I am wondering: when did he update the firmware?  What firmware rev is he using? What is the diff between version 3.5 and 4?

I don't have one of those Xiaomi device, so I can't test anything.

Cheers

Ernst79, 2021-07-26:
I'm not sure if all the failing data are ATC1441 format, I just took the last one to "study" what is wrong. They seem to be all from different devices. and also some in the "extended format" and some others not. 

```
failing data: 043e4e02020300ae87bb59af011918ff4c000713010e2071aa37320010006164ff000000000000ae0001efa3f561b3761f02011a1bff4c000c0e00afe8e025efb044dbb3995e35da10065b1e5c501877bc
failing data: 043e4602020000eef0875da17c1109ff89004591624583320302fefe020104a80400eef0875da17c1f0309474e1aff89005504001f0111001a0d11064ad365128aac94fc68448160a8
failing data: 043e3c020202015b698bbad1471c03039ffe17169ffe0000000000000000000000000000000000000000bc04015b698bbad1470a09ffe000b213ca8d9b0fba
failing data: 043e2702020001962fda3146681102011a020a0c0aff4c0010050318f88918ba0401962fda31466800ba
```

The diff between version 3.5 and 4 is probably the BLE monitor version number, as we are now on 4.x. But I didn't change anything on the aioblescan part, so I doubt that has anything to do with it. I also haven't heard any complains from other users. I suspect a firmware issue to be honest. It looks like the advertisements are just messed up of for reason. 

@LEJOUI could you answer the other questions, especially about whether you did or not did a firmware update?

Anyway, I guess a try except in BLE monitor would help, to just ignore these advertisements. I just was curious about how this can happen in the first place.

LEJOUI, 2021-07-26:
> I am wondering: when did he update the firmware? What firmware rev is he using? What is the diff between version 3.5 and 4?

Hi!

Unfortunately I’m not sure when I flashed the thermometers an which rev they’re actually running, but the last one was around march/april this year. They have the `ATC1441` firmware. The `3.5` and `4.0` is indeed regarding the BLE monitor version.

frawau, 2021-07-27:
Maybe you can tell me the model of the Xiaomi thermometer you use. I'll get one and check it out... May take a little time...

Ernst79, 2021-07-27:
I doubt that is is related to a specific sensor, I think it is something on the dongle/receiving side. The 4 failing advertisements from different devices (different macs) and are not in ATC format (ATC format should contain `uuid16 == 0x181A`), as far as I can see

Anyway, ATC sensors are 99% of the time LYWSD03MMC, flashed with ATC firmware

LEJOUI, 2021-07-27:
> Maybe you can tell me the model of the Xiaomi thermometer you use. I'll get one and check it out... May take a little time...

Yes, LYWSD03MMC

frawau, 2021-07-29:
Well, got an LYWSD03MMC, flashed it with version 3.1 from pvvx... And I have no problem with aioblescan. Tried on an Intel box running Linux 5.11 and a RPi 4 running Linix 5.10. Works fine using the atc decoding.

Temperature info {'mac address': 'a4:c1:38:6e:65:21', 'temperature': 26.0, 'humidity': 46, 'battery': 100, 'battery_volts': 3.154, 'counter': 33, 'rssi': -57}
Temperature info {'mac address': 'a4:c1:38:6e:65:21', 'temperature': 26.1, 'humidity': 46, 'battery': 100, 'battery_volts': 3.156, 'counter': 35, 'rssi': -54}
Temperature info {'mac address': 'a4:c1:38:6e:65:21', 'temperature': 26.3, 'humidity': 45, 'battery': 100, 'battery_volts': 3.159, 'counter': 37, 'rssi': -56}

Sorry but I don't think the problem is with aioblescan.

Try to change your BT dongle.  Maybe that'll help.

Ernst79, 2021-08-02:
I agree, I think something is wrong with the hardware/firmware or perhaps the Operating System.

frawau, 2021-09-04:
FWIW I saw that problem occur. I was using Ubuntu server 21.04 on a Raspberry Pi 4, and I saw that problem occur. It happened only once and after I restarted aioblescan, the problem was gone, and I have not seen it since.  

I am closing this issue, since it really seems like a OS/Firmware/HW issue.

closed 2021-09-04 by frawau