frawau/aioblescan #37: Extended scan filters duplicates

prensing, 2021-08-05:
I was testing the code for the first time against an ATC thermometer, and only getting 1 packet, yet I could get it every time right near the start. I eventually realized that the default code was picking the Extended scan and that is set to filter duplicates. The non-extended shows all.

Definitely not good to have that inconsistency.

frawau, 2021-08-08:
I am not entirely sure I understand your point. ( I did not write the ATC thermometer plugin)  When I run the ATC prugin, I can see  some info. If I use the raw mode and fillter the MAC address, I can see more packets including 2 different types of service data uuid,   fe:95 and 18:1a. As far as I can tell, the ATC plugin only decodes one of them. Where do you see an inconsistency?

Ernst79, 2021-08-08:
fe:95 is Xiaomi MiBeacon format, 18:1a is ATC advertisement format. ATC firmware can send both at the same time.

prensing, 2021-08-08:
My apologies: my initial report clearly did not describe the issue correctly. The ATC thermometer decode is working fine. 

The issue/inconsistency is in the main program aioblescan, specifically routine BLEScanRequester.send_scan_request(). If it does *not* use the extended scan, then it passes "False" to HCI_Cmd_LE_Scan_Enable, indicating to not remove duplicates. However, if it does use the extended scan, then the current code uses (line 1808):
               command = HCI_Cmd_LE_Set_Extended_Scan_Enable(True, 1)
But "1" means to remove duplicates. To not remove dups, I believe it should be "0".

PR #38 (2021-08-14): Do not filter duplicates
Fixes #37 

removes filtering of duplicates with extended advertisement format (like with the normal advertisements)
closed 2021-08-14 by frawau