frawau/aioblescan #39: How can I get the source code of latest relase?

allenzt, 2021-09-03:
Hi @frawau 

Thanks for your excllent work! I can easily use it to process BLE advertisement.

In this repo, I only can find the latest release is 0.2.4, but in PyPi, the latest version is 0.2.8, so could you help me to figure out which one I should use? how can I get source code of latest relase?

Thanks again!

frawau, 2021-09-04:
Hi,

The sources in this github repo  are the latest ones. Look int setup.py and  aioblescan/__init__.py   It is 0.2.8   What make you think it is 0.2.4?

Ernst79, 2021-09-04:
I think he refers to the "releases" on the main page on github

https://github.com/frawau/aioblescan/releases

frawau, 2021-09-04:
I see. That is because I changed the way I push releases to PyPi.  I now use twine to push my releases, so I no longer need to create those "release" bundles.  @allenzt Just clone the repo and you'll get the latest version. You can also download it as a zip. @Ernst79 Thanks for the pointer.

allenzt, 2021-09-04:
@frawau @Ernst79 Many thanks for your help, I got it!

closed 2021-09-04 by allenzt