frawau/aioblescan #40: struct.error: unpack requires a buffer of 2 bytes

openedhardware, 2021-09-20:
Installed on RPi 4 and got following error:
```
pi@raspberrypi:~ $ sudo python3 -m aioblescan -e

Google Beacon {'tx_power': -24, 'url': 'https://www.minew.com/', 'rssi': -68, 'mac address': 'ac:23:3f:85:2a:69'}
Fatal error: protocol.data_received() call failed.
protocol: <aioblescan.aioblescan.BLEScanRequester object at 0xb60b0630>
transport: <_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/selector_events.py", line 813, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.7/dist-packages/aioblescan/aioblescan.py", line 1852, in data_received
    self.process(packet)
  File "/usr/local/lib/python3.7/dist-packages/aioblescan/__main__.py", line 64, in my_process
    xx = EddyStone().decode(ev)
  File "/usr/local/lib/python3.7/dist-packages/aioblescan/plugins/eddystone.py", line 368, in decode
    data = myinfo.decode(data)
  File "/usr/local/lib/python3.7/dist-packages/aioblescan/aioblescan.py", line 703, in decode
    self.val = unpack(">h", data)[0] / 256.0
struct.error: unpack requires a buffer of 2 bytes
```

Next time:
```
pi@raspberrypi:~ $ sudo python3 -m aioblescan -e
Google Beacon {'tx_power': -24, 'name space': b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\n', 'instance': b'\x00\x00\x00\x00\x00\x90', 'rssi': -59, 'mac address': 'ac:23:3f:85:2a:68'}
Google Beacon {'tx_power': -24, 'url': 'https://www.minew.com/', 'rssi': -62, 'mac address': 'ac:23:3f:85:2a:69'}
Fatal error: protocol.data_received() call failed.
protocol: <aioblescan.aioblescan.BLEScanRequester object at 0xb60ae630>
transport: <_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/selector_events.py", line 813, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.7/dist-packages/aioblescan/aioblescan.py", line 1852, in data_received
    self.process(packet)
  File "/usr/local/lib/python3.7/dist-packages/aioblescan/__main__.py", line 64, in my_process
    xx = EddyStone().decode(ev)
  File "/usr/local/lib/python3.7/dist-packages/aioblescan/plugins/eddystone.py", line 368, in decode
    data = myinfo.decode(data)
  File "/usr/local/lib/python3.7/dist-packages/aioblescan/aioblescan.py", line 703, in decode
    self.val = unpack(">h", data)[0] / 256.0
struct.error: unpack requires a buffer of 2 bytes
```

Python version and installed library version:
```
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import aioblescan
>>> aioblescan.__version__
'0.2.8s'
>>> 
```

Instal

frawau, 2021-09-20:
Hi,

What is the EddyStone Beacon hardware you are scanning?

Can you get the raw packets for that MAC address?

Looks like URL and UID are working. It could be a problem with telemetry or eid.

It could also be related to [this problem](https://github.com/frawau/aioblescan/issues/36#issue-952330827)

Cheers

openedhardware, 2021-09-21:
@frawau 

We are using couple of beacons:
https://www.minew.com/product/e8-asset-tag/
https://www.minew.com/product/s2-light-sensor/

Thanks!

frawau, 2021-09-25:
Hi,

It would take me time and money to get one of those. I am not sure I want to buy something I am pretty sure I will never use. The raw output would really help.

Something like
    aioblescan --eddy --raw --mac ac:23:3f:85:2a:69

openedhardware, 2021-09-25:
They are using their custom data structure. Just received a document and I may have to parse myself.

Thanks!

closed 2021-09-25 by openedhardware