frawau/aioblescan #42: Error on decoding TLM Eddystone Beacons

ts-kris, 2021-10-29:
While testing with `python3 -m aioblescan -e` I get the following error upon a TLM beacon:
```
Fatal error: protocol.data_received() call failed.
protocol: <aioblescan.aioblescan.BLEScanRequesize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 870, in _read_ready__data_received
  File "/usr/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 1627, in data_received
  File "/usr/lib/python3.9/site-packages/aioblescan/__main__.py", line 65, in my_process
  File "/usr/lib/python3.9/site-packages/aioblescan/plugins/eddystone.py", line 368, in decode
  File "/usr/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 701, in decode
struct.error: unpack requires a buffer of 2 bytes
```

I don't readily speak Python but I was able to track it down to these lines:
https://github.com/frawau/aioblescan/blob/5fcb8f41af04bd47893b51f40c08ef2980ae141b/aioblescan/plugins/eddystone.py#L367-L368

It looks like the unpack call in Float88 doesn't think 2 bytes are available for it. If I change https://github.com/frawau/aioblescan/blob/5fcb8f41af04bd47893b51f40c08ef2980ae141b/aioblescan/plugins/eddystone.py#L367 to use `aios.ShortInt` instead of `Float88`, aioblescan correctly gets data from TLM beacons (but its not correctly cast).

I'm not sure what the correct fix is here, but I would like to see this fixed as I'm looking to use aioblescan for TLM beacon decoding.

ts-kris, 2021-10-29:
Looks like this is what was intended? (I compared it to ShortInt.decode)
```
diff --git a/aioblescan/aioblescan.py b/aioblescan/aioblescan.py
index 22feb07..7c21d41 100644
--- a/aioblescan/aioblescan.py
+++ b/aioblescan/aioblescan.py
@@ -700,7 +700,7 @@ class Float88:
         return val
 
     def decode(self, data):
-        self.val = unpack(">h", data)[0] / 256.0
+        self.val = unpack(">h", data[:2])[0] / 256.0
         return data[2:]
 
     def __len__(self):

frawau, 2021-10-30:
You are correct.   Thank for pointing me at the precise line./

I pushed the fix, can you test it before I release a new version please?

frawau, 2021-10-30:
Also... Can you tell me what hardware you are using.

ts-kris, 2021-11-01:
The latest commit builds/installs/runs fine.

I'm testing on a TS-7553-V2 (i.MX6UL based platform) with both a generic CSR USB BT dongle and an embedded WILC3000 module from Microchip.

Thanks for the quick turn on this!

frawau, 2021-11-03:
Pushed 0.2.9 to pypi. If everything is OK, please close this issue.

Device looks nice, but a bit expensive just for testing ;)

ts-kris, 2021-11-03:
@frawau Oh, I may have misunderstood. For the Eddystone beacons I'm using both a RSL10-coin-gevb and a "beacon simulator" app in Android.

(I'll leave this open another day or so just to ensure you've seen the message. Will test the pypi release ASAP)

closed 2021-11-05 by ts-kris