frawau/aioblescan #44: 'NoneType' object is not subscriptable

steals, 2022-01-06:
Hello,

I'm getting `'NoneType' object is not subscriptable` error when run `python3 -m aioblescan` on Debian Bullseye (armv6, Rapberry Pi 1 B+)
Python version: 3.9.2
aioblescan: 0.2.10

Here is the stacktrace:
```
Fatal error: protocol.data_received() call failed.
protocol: <aioblescan.aioblescan.BLEScanRequester object at 0xb61293e8>
transport: <_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 870, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1852, in data_received
    self.process(packet)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/__main__.py", line 51, in my_process
    xx = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1303, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1350, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 833, in decode
    data = field.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1473, in decode
    data = x.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 236, in decode
    self._val = unpack(">B", data[:1])[0]
TypeError: 'NoneType' object is not subscriptable
```

frawau, 2022-01-07:
Hi,
Would you have any of the device with a plugin? Ruuvi, Mi Thermomether? Do you know what devices you have around you? Can you run aioblescan with raw output?
   sudo aioblescan -R

So I might get some idea of what is wrong.

I do not have a RPi 1, only 0,2,3,4...  I won't be able to test on that platform

steals, 2022-01-07:
Yes, I have Xiaomi thermometer LYWSD03MMC

Here is the raw log:
```
Raw data: b'\x04\x0e\x04\x02A \x0c'
HCI Event:
    code:
        0e
    length:
        4
        allow pkt:
            2
        Cmd Group:
            b'\x08'
        Cmd Code:
            b'A'
        resp code:
            0c
Raw data: b'\x04\x0e\x04\x02B \x0c'
HCI Event:
    code:
        0e
    length:
        4
        allow pkt:
            2
        Cmd Group:
            b'\x08'
        Cmd Code:
            b'B'
        resp code:
            0c
Fatal error: protocol.data_received() call failed.
protocol: <aioblescan.aioblescan.BLEScanRequester object at 0xb6049430>
transport: <_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 870, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1852, in data_received
    self.process(packet)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/__main__.py", line 51, in my_process
    xx = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1303, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1350, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 833, in decode
    data = field.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1473, in decode
    data = x.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 236, in decode
    self._val = unpack(">B", data[:1])[0]
TypeError: 'NoneType' object is not subscriptable
```

frawau, 2022-01-07:
The raw data is rather short. It seems the packet triggering the error is not shown. 

Does it crash if you run
   sudo aioblescan -A

?
Last time I saw a problem like that, I think it turned out to be a driver problem or some bad BLE device.   Can you also try on some different computer? 

Also, if you know the MAC address of the Mi Thermometer, try to run

    sudo aioblescan -A -m <mac address>

steals, 2022-01-07:
_Does it crash if you run sudo aioblescan -A_ - Yes

If I run with specific mac address I don't see any raw data at all
`root@DietPi:~# python3 -m aioblescan -A -m A4:C1:38:B2:24:97 -R`

> root@DietPi:~# python3 -m aioblescan -A -m A4:C1:38:B2:24:97 -R
Fatal error: protocol.data_received() call failed.
protocol: <aioblescan.aioblescan.BLEScanRequester object at 0xb60ac3d0>
transport: <_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 870, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1852, in data_received
    self.process(packet)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/__main__.py", line 51, in my_process
    xx = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1303, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1350, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 833, in decode
    data = field.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1473, in decode
    data = x.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 236, in decode
    self._val = unpack(">B", data[:1])[0]
TypeError: 'NoneType' object is not subscriptable
^Ckeyboard interrupt
closing event loop

frawau, 2022-01-07:
With no -R option?

steals, 2022-01-07:
It crashes almost immediately within 5 sec after the start

frawau, 2022-01-07:
Grrr .... Difficult to figure things out with so little data.... 

The last time I saw this, the packets were malformed. It was announcing data of a certain length, but the data was not in the packet (hence None). After further investigation, it turned out the Bluetooth device driver was messing up the received packets. Running with a different environment solved the problem.

You think you can hack the code a bit to get more info? Just a print line before any processing.

ukBaz, 2022-01-07:
As a suggestion for debugging... Have `sudo btmon` running in a different terminal. That should give you the raw data coming in.

steals, 2022-01-07:
_You think you can hack the code a bit to get more info? Just a print line before any processing._ - Yes, I can, please suggest the lines
Will also capture the btmon output now.

steals, 2022-01-07:
here is the btmon log started rigth before aioblescan and stopped right after crash. https://pastebin.com/QGU6zbGv

steals, 2022-01-07:
Also added two print statements in __main__.py (line 50) here is the output:
```
python3 -m aioblescan -A
*** my_process start ***
b'\x04\x0e\x04\x02A \x0c'
*** my_process start ***
b'\x04\x0e\x04\x02B \x0c'
*** my_process start ***
b"\x04>3\r\x01\x13\x00\x01Q\x04\xa84'\xdd\x01\x00\xff\x7f\xb2\x00\x00\x00\x00\x00\x00\x00\x00\x00\x19\t\tRK-G211S\x02\x01\x06\x0b\xff\xc2\x01\x00\x00Q\x04\xa84'\xdd"
*** my_process start ***
b"\x04>3\r\x01\x13\x00\x01Q\x04\xa84'\xdd\x01\x00\xff\x7f\xb4\x00\x00\x00\x00\x00\x00\x00\x00\x00\x19\t\tRK-G211S\x02\x01\x06\x0b\xff\xc2\x01\x00\x00Q\x04\xa84'\xdd"
*** my_process start ***
b'\x04>5\r\x01\x13\x00\x00,\xe5y&\xc6\x88\x01\x00\xff\x7f\xb8\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1b\x02\x01\x1a\x02\x08\x00\x03\x03a\xfe\x10\xff\x03\x00\x00e\xe4\x00\x00\x0f\x9c\xf3\x87\xc1u\xcf2'
*** my_process start ***
b'\x04>(\r\x01\x13\x00\x01<\xf9\xad:o`\x01\x00\xff\x7f\xb2\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0e\x02\x01\x1a\x02\n\x0c\x07\xffL\x00\x10\x02\x03\x00'
*** my_process start ***
b'\x04>)\r\x01\x10\x00\x01\\)\xbb\x82K\t\x01\x00\xff\x7f\xbe\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x02\x01\x1a\x0b\xffL\x00\t\x06\x03&\xac\x10\x01e'
*** my_process start ***
b'\x04>,\r\x01\x13\x00\x01\x8a\x86c\xfb\x11y\x01\x00\xff\x7f\xb2\x00\x00\x00\x00\x00\x00\x00\x00\x00\x12\x02\x01\x1a\x02\n\x08\x0b\xffL\x00\x10\x06%\x1a(\xa2\xa4\xbb'
*** my_process start ***
b"\x04>3\r\x01\x13\x00\x01Q\x04\xa84'\xdd\x01\x00\xff\x7f\xb4\x00\x00\x00\x00\x00\x00\x00\x00\x00\x19\t\tRK-G211S\x02\x01\x06\x0b\xff\xc2\x01\x00\x00Q\x04\xa84'\xdd"
*** my_process start ***
b"\x04>f\r\x02\x13\x00\x00,\xe5y&\xc6\x88\x01\x00\xff\x7f\xb6\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1b\x02\x01\x1a\x02\x08\x00\x03\x03a\xfe\x10\xff\x03\x00\x00ed\x00\x00\x0f\x9c\xf3\x87\xc1u\xcf2\x13\x00\x01Q\x04\xa84'\xdd\x01\x00\xff\x7f\xb0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x19\t\tRK-G211S\x02\x01\x06\x0b\xff\xc2\x01\x00\x00Q\x04\xa84'\xdd"
Fatal error: protocol.data_received() call failed.
protocol: <aioblescan.aioblescan.BLEScanRequester object at 0xb60d94a8>
transport: <_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 870, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1852, in data_received
    self.process(packet)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/__main__.py", line 52, in my_process
    xx = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1303, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1350, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 833, in decode
    data = field.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1473, in decode
    data = x.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 236, in decode
    self._val = unpack(">B", data[:1])[0]
TypeError: 'NoneType' object is not subscriptable
^Ckeyboard interrupt
closing event loop
```

frawau, 2022-01-07:
Thanks. I'll have a look at that tomorrow.... It is evening here and I had a long trip yesterday (including an 11hrs flight)

frawau, 2022-01-09:
I did have a look. I can see some packets that parse correctly, and one that does not. One think is that both packets contain the string "RK-G211S" which I think is your Redmond SkyKettle RK-G211S.  You can verify that the packet that fails to parse comes from that device by unplugging it and running aioblescan afterward.

I did notice that the failing packet contains 2 "Extended Advertising reports" and that the length of the payload is exactly twice that of an OK packet (102 vs 51) yet the content of the first ext adv report seems a bit strange (name, manufacturer id,...)  and it consumes so much data that there is not enough left for the second one.

Looks like the packet not formatted properly, but I will need to go back into the BLE specs to see what exactly. I have not done that for quite a while, this will take some time

steals, 2022-01-09:
Didn't know my kettle sends something, lol %)
I've turned it off and still have the same issue. Here are a few more btmon logs https://pastebin.com/nwEKVeAP https://pastebin.com/BxJtgcns  https://pastebin.com/92KzfNUn  https://pastebin.com/KVEFGGMw

frawau, 2022-01-10:
The error is similar. 2 Ext Adv Report and it runs out data in the second. Starts to look like the lower level is messing up the packets it receives. When I can get time from work, I 'll see if I can install Bullseye on RPi2 and see if I can reproduce it. On your side, if you have another Linux machine lying around you could try with that.

shadowsgit, 2022-01-10:
oops, didn't  read all comments in that issue... But looks me just filled trivial repro case for same issue and point to simple fix #45

frawau, 2022-01-13:
I will include @shadowsgit fix and publish a new version soon

Ernst79, 2022-01-14:
@frawau, @shadowsgit I have included aioblescan 0.2.11 in [BLE monitor (7.0.0)](https://github.com/custom-components/ble_monitor) and have asked the users with a similar error to check. @LEJOUI is one of these users, but he mentioned that the error still exists. The only difference I notice it that he is not using the Extended format. However, in that part of the code, I do see a `return data`, see https://github.com/frawau/aioblescan/blob/7031b9f2aada90f40434b312814cd3f7296351f4/aioblescan/aioblescan.py#L1352

```
Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:98 
First occurred: 13 januari 2022 23:54:43 (2 occurrences) 
Last logged: 05:00:48

Error doing job: Fatal error: protocol.data_received() call failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/selector_events.py", line 870, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 1841, in data_received
    extra_data = ev.decode(packet)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 1303, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 1350, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 833, in decode
    data = field.decode(data)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 1375, in decode
    data = x.decode(data)
  File "/usr/local/lib/python3.9/site-packages/aioblescan/aioblescan.py", line 191, in decode
    self.val = unpack(">B", data[:1])[0]
struct.error: unpack requires a buffer of 1 bytes
```

In the past, I have received some failing data from @LEJOUI, which I also posted in #36. Back then, @frawau concluded that it was most likely a hardware/firmware issue. We might still conclude this, but I thought it's worth to ask you guys if you still think this is just an hardware issue? 

```
failing data: 043e4e02020300ae87bb59af011918ff4c000713010e2071aa37320010006164ff000000000000ae0001efa3f561b3761f02011a1bff4c000c0e00afe8e025efb044dbb3995e35da10065b1e5c501877bc
failing data: 043e4602020000eef0875da17c1109ff89004591624583320302fefe020104a80400eef0875da17c1f0309474e1aff89005504001f0111001a0d11064ad365128aac94fc68448160a8
failing data: 043e3c020202015b698bbad1471c03039ffe17169ffe0000000000000000000000000000000000000000bc04015b698bbad1470a09ffe000b213ca8d9b0fba
failing data: 043e2702020001962fda3146681102011a020a0c0aff4c0010050318f88918ba0401962fda31466800ba
```

shadowsgit, 2022-01-14:
@Ernst79  me also used its as HA integration, but in my case it somehow works fine without above fix with HA, but it was failed in few seconds when I launch aioblescan as is. (or HA at least restart plugin). To be honest I don't know how to get similar log from HA as mentioned in https://github.com/custom-components/ble_monitor/issues/433 ))
Formally you reported other issue... But looking into official [bluez sample](https://github.com/bluez/bluez/blob/9ee79f67bed2e733569c483b9ef4440710448ad0/tools/parser/hci.c#L3596) make me believe that payload inside #36 is totally correct and it pairing should look like:
```
--------------------------------------------------------------------------------------------------------------------------------------------------------------
HCI  Evt Len Sub Num Evt  Peer -------MAC-------   Len  Len Type Val    Len Type Val   Len Type -------Val---------------- RSSI--Evt  Peer -------MAC-------        RSSI
type code    evt rep type addr                              flag            flag           flag
--------------------------------------------------------------------------------------------------------------------------------------------------------------
 04   3e 27  02  02   00   01  96 2f da 31 46 68   11   02   01  1a     02   0a  0c     0a  ff  4c 00 10 05 03 18 f8 89 18   ba   04   01   96 2f da 31 46 68 00    ba

```

So repro code

```python
import aioblescan as aiobs
ev = aiobs.HCI_Event()
data=bytearray.fromhex("043e2702020001962fda3146681102011a020a0c0aff4c0010050318f88918ba0401962fda31466800ba")
xx = ev.decode(data)
ev.show(0)
```

And the fix #46:

Ernst79, 2022-01-14:
@shadowsgit Thanks, I can confirm that #46 seems to fix it, all 4 packets now pass nicely. I'll wait for @frawau, when he makes a new release, I will update it in BLE monitor. Thanks to you both @shadowsgit and @frawau for the excelent work. 

BTW @shadowsgit. In HA, I normally add the following line in BLE monitor to get the data at the place where I need it, when I am debugging the code. 

```python
_LOGGER.error("failing data: %s",data.hex()")` 
```

frawau, 2022-01-15:
@shadowsgit  Thanks for the PR
@Ernst79  new release pushed to PyPi

steals, 2022-01-15:
I can confirm the latest 0.2.12 version is not crashing anymore on my side. Thank you @shadowsgit  and @frawau

closed 2022-01-15 by steals