frawau/aioblescan #48: Cannot scan when device is bluetooth 5

maripo31, 2022-03-01:
Hello,
When device is bluetooth 5, the function self._use_ext_scan() return True.
Then HCI_Cmd_LE_Set_Extended_Scan_Enable is actived.
But the normal le scan is not actived and no beacon 4.0 message was received.

My suggestion is in send_scan_request to remove the else: and make always the 2 functions :
HCI_Cmd_LE_Set_Scan_Params and
HCI_Cmd_LE_Scan_Enable

Here is the patch 
*** aioblescan.py       2022-03-01 17:36:54.794613192 +0000
--- aioblescan.py.old   2022-03-01 17:32:37.587643554 +0000
*************** class BLEScanRequester(asyncio.Protocol)
*** 1809,1821 ****
              self._send_command_no_wait(command)

              command = HCI_Cmd_LE_Set_Extended_Scan_Enable(True, 0)
              self._send_command_no_wait(command)
-         sparam = int(isactivescan)
-         command = HCI_Cmd_LE_Set_Scan_Params(scan_type=sparam)
-         self._send_command_no_wait(command)

!         command = HCI_Cmd_LE_Scan_Enable(True, False)
!         return self._send_command_no_wait(command)

      async def stop_scan_request(self):
          """Sending LE scan request"""
--- 1809,1822 ----
              self._send_command_no_wait(command)

              command = HCI_Cmd_LE_Set_Extended_Scan_Enable(True, 0)
+             return self._send_command_no_wait(command)
+         else:
+             sparam = int(isactivescan)
+             command = HCI_Cmd_LE_Set_Scan_Params(scan_type=sparam)
              self._send_command_no_wait(command)

!             command = HCI_Cmd_LE_Scan_Enable(True, False)
!             return self._send_command_no_wait(command)

      async def stop_scan_request(self):
          """Sending LE scan request"""

Regards.

Magalex2x14, 2022-03-01:
Are you saying that aioblescan doesn't work for you in its current form? This is strange. With extended scanning, messages from BT4.0 devices are received without any problems, they just come in an extended format. If  _use_ext_scan flag is set, then adapter supports extended scanning.

Can you tell us more about your situation? What adapter, what kernel version, and what Linux distro you use?
Have you tried diagnosing a failure of extended scan request (look at `btmon` output, for example)?

I'm one of the developers of a home automation system component using aioblescan. At the moment, I am not aware of any cases of such a problem. I personally use both the BT4.0 and BT5.0 adapters (RTL8761B chip) in one system at the same time, and aioblescan works excellent on it, initiating a legacy scan on first adapter and an extended scan on the second.
For example, LYWSDCGQ Bluetooth 4.1 device extended advertisement report:
```
> HCI Event: LE Meta Event (0x3e) plen 51                                                                                                                         #5910 731.392562
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0013
            Props: 0x0013
              Connectable
              Scannable
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: ADV_IND (0x0013)
          Address type: Public (0x00)
          Address: 58:2D:34:31:40:00 (Qingping Electronics (Suzhou) Co., Ltd)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -81 dBm (0xaf)
          Periodic advertising invteral: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x19
        02 01 06 15 16 95 fe 50 20 aa 01 f1 00 40 31 34  .......P ....@14
        2d 58 0d 10 04 ec 00 46 01                       -X.....F.
```

Magalex2x14, 2022-03-01:
In addition, nothing prevents you from initiating a scan "manually", that is, by executing the send_command or _send_command_no_wait methods with arbitrary commands instead of calling send_scan_request.

But personally it would be more interesting to me to understand the reasons of the behavior described by you.