frawau/aioblescan #49: Early return from decoding Eddystone TLM beacon decode loses RSSI and MAC info in packet

ts-kris, 2022-05-02:
I noticed that in aioblescan, all Eddystone beacons also get rssi and mac appended to them in their .decode() function, but not TLM.

e.g.
https://github.com/frawau/aioblescan/blob/master/aioblescan/plugins/eddystone.py#L305-L306 UID beacon does not return
https://github.com/frawau/aioblescan/blob/master/aioblescan/plugins/eddystone.py#L358 URL beacon does not return
https://github.com/frawau/aioblescan/blob/master/aioblescan/plugins/eddystone.py#L385-L386 EID beacon does not return
https://github.com/frawau/aioblescan/blob/master/aioblescan/plugins/eddystone.py#L379 TLM returns and misses out on running the rssi and mac lines before final function return:

https://github.com/frawau/aioblescan/blob/master/aioblescan/plugins/eddystone.py#L387-L393

I'm not sure if this is intended or desired or what, but, I need it in my application. A quick patch removing the return gets me rssi and mac in the final packet decode
```
diff --git a/aioblescan/plugins/eddystone.py b/aioblescan/plugins/eddystone.py               
index 214cdd5..0136a4e 100644                                                                
--- a/aioblescan/plugins/eddystone.py                                                        
+++ b/aioblescan/plugins/eddystone.py                                                        
@@ -376,7 +376,6 @@ class EddyStone(object):                                                 
             data = myinfo.decode(data)
             found.payload.append(myinfo)
             result["uptime"] = myinfo.val * 100  # in msecs
-            return result
         # elif etype.val== ESType.tlm.eid:
         else:
             result["data"] = data
```

frawau, 2022-05-04:
Thanks for the report and fix.

Willl release soon.

frawau, 2022-05-04:
Just curious, what is your application?

frawau, 2022-05-04:
Try it and if it is OK, please close this issue.

ts-kris, 2022-05-04:
> Just curious, what is your application?
Its a demo for using some of our hardware to capture TLM beacon information. The fix is important because without it, there is no indication of specifically what hardware sent the beacon out. e.g. If there are 5 beacons reporting 5 different temperatures and time stamps, there was no way to tell which was which.

Despite Eddystone being more or less discontinued as intended by Google, many BLE beacon EVKs implement Eddystone still out of the box.

> Try it and if it is OK, please close this issue.
Looks good to me!