frawau/aioblescan #50: aioblescan can't find HCI device

Nighthawk79, 2022-05-13:
Hello,

since i've updated my Ubuntu from 21.10 to 22.04, aiovlescan is not functional anymore.

The error i'm getting is 

`#aioblescan -m 50:FB:19:4E:77:7C
Traceback (most recent call last):
  File "/usr/local/bin/aioblescan", line 33, in <module>
    sys.exit(load_entry_point('aioblescan==0.2.13', 'console_scripts', 'aioblescan')())
  File "/usr/local/lib/python3.10/dist-packages/aioblescan-0.2.13-py3.10.egg/aioblescan/__main__.py", line 176, in main
    mysocket = aiobs.create_bt_socket(opts.device)
  File "/usr/local/lib/python3.10/dist-packages/aioblescan-0.2.13-py3.10.egg/aioblescan/aioblescan.py", line 1760, in create_bt_socket
    raise exceptions[0]
OSError: [Errno 19] error while attempting to bind on interface 0: No such device`

The HCI Device is functional:

`#hcitool dev
Devices:
	hci1	E8:48:B8:C8:20:00`

and for example hcitool lescan is working properly:

`#sudo hcitool lescan
LE Scan ...
18:93:7F:F2:AD:42 (unknown)
18:93:7F:F2:AD:42 Somfy Outdoor 0076B10826E4
00:7C:2D:C3:E3:9F (unknown)
61:0C:47:C5:90:58 (unknown)
54:D2:72:79:44:0C (unknown)
61:0C:47:C5:90:58 nut
50:FB:19:4E:77:7C MIBFS
61:0C:47:C5:90:58 (unknown)
50:FB:19:4E:77:7C (unknown)
50:FB:19:4E:77:7C MIBFS
F0:50:63:D4:E4:11 (unknown)
F0:50:63:D4:E4:11 Mi Smart Band 5
32:24:A9:6C:2A:72 (unknown)
32:24:A9:6C:2A:72 (unknown)`

`#dmesg | grep Blue 
[    8.412679] Bluetooth: Core ver 2.22
[    8.412714] Bluetooth: HCI device and connection manager initialized
[    8.412720] Bluetooth: HCI socket layer initialized
[    8.412724] Bluetooth: L2CAP socket layer initialized
[    8.412730] Bluetooth: SCO socket layer initialized
[    8.530612] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[    8.531562] Bluetooth: hci0: RTL: rom_version status=0 version=1
[    8.531567] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_fw.bin
[    8.533610] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_config.bin
[    8.533804] Bluetooth: hci0: RTL: cfg_sz 6, total sz 27814
[    8.675581] Bluetooth: hci0: RTL: fw version 0x09a98a6b
[   11.074229] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   11.074235] Bluetooth: BNEP filters: protocol multicast
[   11.074242] Bluetooth: BNEP socket layer initialized
[   12.795945] Bluetooth: RFCOMM TTY layer initialized
[   12.795955] Bluetooth: RFCOMM socket layer initialized
[   12.795964] Bluetooth: RFCOMM ver 1.11
[50015.919166] Bluetooth: hci1: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[50015.920153] Bluetooth: hci1: RTL: rom_version status=0 version=1
[50015.920161] Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_fw.bin
[50015.920309] Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_config.bin
[50015.920382] Bluetooth: hci1: RTL: cfg_sz 6, total sz 27814
[50016.059153] Bluetooth: hci1: RTL: fw version 0x09a98a6b`

How can i solve the problem?

frawau, 2022-05-13:
You really should read the error message.

    It shows  "interface 0: No such device"

And when you show hcitool works  it shows
     hcitool dev Devices: hci1 E8:48:B8:C8:20:00

 0 == hci0 vs hci1

If you had used 
    aioblescan -h

You would have seen 
    -D DEVICE, --device DEVICE
                        Select the hciX device to use (default 0, i.e. hci0).

Try 
    -D 1

Nighthawk79, 2022-05-14:
you are absolutely right.
unfortunately  if i strat aioblescan -D 1, i don't get any messages.

frawau, 2022-05-14:
It works for me on Kubuntu 22.04.  Make sure you run as root to avoid permission issues
     sudo aioblescan -D 1 ....

Nighthawk79, 2022-05-14:
oh, got it now, it works with root.

closed 2022-05-14 by Nighthawk79