frawau/aioblescan #53: Process exits immediately on Intel OSX

dankozlowski, 2023-01-13:
I'm trying to run this library on my 2020 Macbook Pro (Intel).  The process exits as soon as it runs, and I'm not able to enable any verbosity so far as I can tell to further troubleshoot.  

Does this work on Intel hardware?  Apologies if these questions are naive.

```
$ pip3 install 
$ sudo python3 -m aioblescan -r
Password:
$
$ python3 -m aioblescan -r
$
```

No output, no nada.

frawau, 2023-01-14:
I will have to wait until Monday or later before I can test that on one of my colleagues' Mac, I do not use Mac.

You may need to specify the device  (use -h for help). What does hciconfig reports (if it exists)?

Could be that the kernel/python is not compiled with Bluetooth socket support. 

I usually test on Raspberry Pi 4 with Debian and Ubuntu (Kubuntu actually) with and64

bitshop, 2023-01-23:
Process also exits immediately on Windows it seems, running as admin or not.  Perhaps an error message would be an interim fix? I'll try a Pi another day. If you'd prefer a new issue that is Windows specific let me know and I'll open one.