frawau/aioblescan #54: struct.error: unpack requires a buffer of 2 bytes (ATCMiThermometer with Xiaomi beacon format)

arepih, 2023-07-28:
Installed on RPi 4 and got error below after running a while:
**sudo aioblescan - m xx:xx:xx:xx:xx:xx**
Device is ATCMiThermometer with pvvx firmware v4.3 using Xiaomi beacon format (not ATC format)
I trying to get data from this thermometer both with older mi thermometer (circle shape, stock firmware with Xiaomi beacon format).

```
Fatal error: protocol.data_received() call failed.
protocol: <aioblescan.aioblescan.BLEScanRequester object at 0xf7094c70>
transport: <_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 870, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1855, in data_received
    self.process(packet)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/__main__.py", line 51, in my_process
    xx = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1303, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1350, in decode
    data = ev.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 833, in decode
    data = field.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1379, in decode
    data = ad.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 1649, in decode
    val.decode(data[: length.val - len(type)])
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 664, in decode
    data = mynbyte.decode(data)
  File "/usr/local/lib/python3.9/dist-packages/aioblescan/aioblescan.py", line 623, in decode
    self.val = unpack(">%ds" % self.length, data[: self.length])[0][::-1]
struct.error: unpack requires a buffer of 2 bytes
```

Ernst79, 2023-07-28:
How did you install HA? This error normally means that something is wrong with Bluetooth on the Pi or dongle. Either on hardware level, or on OS level.

arepih, 2023-07-28:
I’m not using HA (Home assistant?).
Actually someone open same issue previously
[struct.error: unpack requires a buffer of 2 bytes #40](https://github.com/frawau/aioblescan/issues/40) 
 and it was because the device use custom data structure so I thought maybe Xiaomi beacon format is not supported.
I will try changing the thermometer to ATC beacon format and check if same error happen or not.

Ernst79, 2023-07-28:
Ah, sorry, my mistake.

arepih, 2023-07-31:
Changed to ATC format, same error.
Move the sd card from rpi4 into rpi3B+, same error.
Remove and reinsert battery from thermometer still error.
Reboot the rpi, uninstall aioblescan then install again, finally no more error.

closed 2023-07-31 by arepih